wgcv/RAWR-TFT-Firmware-Artillery3D #1: [BUG] Filament change (M600) did not work

BenjaminFaal, 2020-07-23:
It did ask me for the filament change i pressed ok, but when i pressed out it did nothing. I think its because the temperature was 200/200 and it was waiting for it to change.

See this video: https://www.youtube.com/watch?v=YwkXddmQImc

wgcv, 2020-07-28:
Sorry did you solved? I think that someone wrote to me internally ask me something like these and finally solved change ina setting on Cura.

BenjaminFaal, 2020-07-28:
I have a workaround fix, when it asks for the filament change i first set the temperature manually to 201 and then filament change in and out always works correctly. Then after that i resume the print.

wgcv, 2020-07-28:
Could be a bug I would check the filament change menu in the next week.

closed 2021-01-23 by wgcv