wgcv/RAWR-TFT-Firmware-Artillery3D #7: [BUG] Extruder Temp to 450°C invalid value

FireN1337, 2020-08-06:
Hey,

At first thank you for that wonderful FW.
I changed the Extruder Temps to max 450 in the config ini.
I use a bmg/mosquito clone.

After setting up the SD and uploading it, the screen shows invalid value 450°C.

I can't set the temperature higher than 275°C.
Is that a Bug or did I do anything wrong?

Thanks in Advance

Greetings :)

wgcv, 2020-09-07:
Did you change in the marlin also to 450°C ?

FireN1337, 2020-09-11:
Yes, heater 0 is configured with 450°C

FireN1337, 2021-02-08:
Hello, 
I hope you are well.

I´ve discovered a new bug.
Altough the heatbed is configured with 130°C in Marlin and 150°C in the RAWR TFT FW, the heatbed wont go further than 120°C. (Tried to print ASA this week).
Sidewinder X1 heatbed should go up to 130°C at least.

Would be cool if you could get those two things fixed in 2021 :)
I would do it myself if I could, but I´m more the Fusion360 and Desgin guy than a coder.

When it works I will reward you with some PP-Money. ;)