wgcv/RAWR-TFT-Firmware-Artillery3D #9: [Q] (Your Question?)

MdelgadoL83, 2020-08-15:
**Your question or details you need. Please describe.**
When i send to print a piece in every change of layer appear a message box with the content Unknown command "M73 ..."

According the marlin code the command is for update the process in the display, but is very annoying

can i disable this feature or is a bug?

**specify the device variant related to your question.**
Artillery Genius
[image]

**Additional context**
I have a original marlin firmware and the last rawr firmware for the TFT

wgcv, 2020-08-15:
The message is because you are sending the command and your marlin doesn't recognize the command, you can avoid to send that command from the slicer or update/change your marlin to have the command. Is not a tft firmware error.

MdelgadoL83, 2020-08-15:
ok i get it, but in the original TFT firmware you just ignore this message so it's a good idea if you can disable or enable this feature.

dragoncity17, 2020-08-23:
Yes very annoying ! 
Can't use because every 2 minutes error message.

I use Prusa Slicer and octoprint.

Can you fix that?

MdelgadoL83, 2020-08-24:
yes, solve it, disabling the option "support remain time" in the tab "printer settings" in prusa slicer

dragoncity17, 2020-08-25:
Ok great.

But I have also :

> G21 in the beginning.

> Unknown command "93"

> Line Number is not Last Line Number

Hum...how can I fix that ?

Thank you

dragoncity17, 2020-08-28:
@wgcv 
Can you remove all message in your future release please ?

I disable sound...but when there is error sometimes "block" print during few second !

Thank you ! :)

wgcv, 2020-08-28:
That is a good idea would be in the next release!

dragoncity17, 2022-03-16:
Hello,
I updated with last update but always this problem, can you fix that ?
Because when there is error sometimes "block" print during few second ....