wgcv/RAWR-TFT-Firmware-Artillery3D #13: [Q] (Your Question?)What are the steps when changing the default settings to a new board?

John-Smith980, 2020-10-17:
I'm trying to change the code to match a TFT 3.5 E3 V3.0 Board, I was trying to follow your directions on how to change the board type, and it seems like all I have to do is change the default env: however when I try to compile it there are issues. What are the steps for changing the code for a TFT 3.5 E3 V3.0?

John-Smith980, 2020-10-17:
I keep getting a "TEXT DISTANCE" problem. No Idea how to go from here. The only change I have made is this: default_envs = BIGTREE_TFT35_E3_V3_0

wgcv, 2021-01-23:
I'm not sure what resolution have the BIGTREE_TFT35_E3_V3_0 but check the files in ./TFT/src/User/Variants/Resolution:
TFT_320X240.h line 18
```
#define TEXT_DISTANCE 24
```
You can try to add to TFT_480X320.h that line but you need to test and get the correct TEXT_DISTANCE for that resolution