wgcv/RAWR-TFT-Firmware-Artillery3D #23: [FR] Add an option or variant for manual mesh bed levelling instead of ABL

BenBergman, 2021-01-16:
**Is your feature request related to a problem? Please describe.**
My printer has a distinct warp to it, so I've added manual mesh bed levelling to Marlin to counteract this. RAWR seems to only support ABL, not MBL (unless I missed it).

**Describe the solution you'd like**
I would love an option (in the config file, settings menu, or even a separate build) that switches out the buttons for ABL to a set of buttons for MBL.

**Describe alternatives you've considered**
I have performed manual bed levelling by manually entering gcode commands into my Octoprint terminal, but that is a bit of a pain. I could similarly add these commands to the custom commands section in the RAWR config, but between each mesh point I still need to move the print head to adjust, which would be much nicer with a proper UI with position readings.

**Additional context**
With Marlin's manual mesh bed leveling, `G29 S1` starts a levelling pattern by homing and moving to the first point. You then move the print head (note: not baby-stepping - rookie mistake that got me good). Then you issue `G29 S2` to move to the next point where you can again move the print head to get the right gap. Keep running `G29 S2` for each point. After adjusting the height at the last point, run `G29 S2` one more time to store the final setup. M500 might be needed to save it, I'm unsure. Each G29 command shows which point you are on and the total number, so this feedback could be used for a progress bar or a wizard.

Ideally the UI for this would have a button to start the process, then you have buttons to raise/lower the print head, a button to adjust the step amount (0.1mm, 0.01mm maybe?), a readout showing the current height, a readout showing the current probe point out of the total, and a button to continue to the next one (or possibly finish if on the last one).

wgcv, 2021-02-09:
https://github.com/wgcv/RAWR-TFT-Firmware-Artillery3D/issues/16

BenBergman, 2021-02-10:
Awesome, thanks.

For anyone else stumbling on this, I ended up switching to this TFT firmware which has a nice interactive mesh editor: https://www.thingiverse.com/thing:4294049