wgcv/RAWR-TFT-Firmware-Artillery3D #26: Wrong conventions for mks tft

wtfiAndrewNelson, 2021-02-09:
### Steps to reproduce

1. Load files onto sd card
2. Insert sd card and turn on X1
3. Does not install, just boots as normal

**Expected behavior**

**Actual behavior**

### Hardware Variant

wgcv, 2021-02-09:
All Artillery use a generic mks tft. I tested on SW X1 v4 

1. Check your sd format
2. Restart from the button your SW x1
3. Check the structure file is like the [readme.md](https://github.com/wgcv/RAWR-TFT-Firmware-Artillery3D/blob/master/README.md#how-to-setup-the-tft-firmware)

wtfiAndrewNelson, 2021-02-09:
This is not what just shipped to me, the structure that you are pointing to is the BTT type structure and what is needed with the mks tft is as follows:

1,According to the content that needs to be updated, copy the following file to the SD card:

 (1)、update the config file：mks_config.txt
 (2)、update images：         mks_pic
 (3)、update font：           mks_font
 (4)、update TFT firmware：   mkstft28.bin
 (5)、update Wifi firmware：  MksWifi.bin

also the images are not .bmp they are .bin

This is what is shipping from Artillery3D as mine just arrived a week ago and this is the link to the firmware on the tft: https://github.com/makerbase-mks/MKS-TFT/tree/master/MKS-TFT2.8-3.2

I guess people don't know about this yet as your firmware is not the first that I've run into in the wrong format for the board.

wgcv, 2021-02-09:
Just follow the instructions.

> All Artillery use a generic mks tft. I tested on SW X1 v4
> 
> 1. Check your sd format
> 2. Restart from the button your SW x1
> 3. Check the structure file is like the [readme.md](https://github.com/wgcv/RAWR-TFT-Firmware-Artillery3D/blob/master/README.md#how-to-setup-the-tft-firmware)

wtfiAndrewNelson, 2021-02-09:
I am not trying to fight with you, but you do not have the correct information, here is the link to the official Artillery tft firmware: https://github.com/artillery3d/sidewinder-x1-tft-firmware

I assume that your information was from an older version but it's not the one that I received last week. Thank you for your help.

closed 2021-02-09 by wgcv