wgcv/RAWR-TFT-Firmware-Artillery3D #30: [BUG] Too long extrusion prevented

codigoneo, 2021-07-04:
### Description

I was printing a keychain with filament color change and after changing the filament I hit continue. Then I got an error "too long extrusion prevented". The printer went crazy and beeped. I tried to continue printing several times but it can not. I tried changing the filament again, but it didn't solve the problem. Finally I tried to turn off the printer and recover the print, but it continued to print in the air (where the filament change was made).

### Steps to reproduce

1. Steps to Reproduce
2. Print the gcode
3. Change filament
4. Resume print

**Expected behavior**
Print normally

**Actual behavior**
An error appears and does not print

### Hardware Variant
Artillery Sidewinder X1 with touch TFT

### TFT Firmware Version & Main Board Firmware details
https://github.com/pinguinpfleger/ASWX1-FW-MOD/releases/tag/ASWX1-FW-MOD-v1.1 without changes
https://github.com/wgcv/RAWR-TFT-Firmware-Artillery3D/releases/tag/V28.0.2 without changes

### Additional Information
* gcode file
[audiKeychain.zip](https://github.com/pinguinpfleger/ASWX1-FW-MOD/files/6759815/audiKeychain.zip)

fkahhaleh, 2021-07-26:
Just wanted to confirm same thing happened with me.
I was able to do one successful filament change, second print failed with the same error.
To get rid of the beeping, I had to press (and hold the press) on the OK button of the dialog window until it stopped.
But any subsequent resume operation attempt failed with the same error/beeping.

Tomtomtom74, 2021-08-04:
Hello,
I have observed the same problem. I noticed when this problem occurs that when I start printing, the Z-zero position is not zero but has a very high nonsense value. I fix this by disconnecting the printer (Sidewinder X1) from the power supply before printing. If the Z-zero position is then zero when the print is restarted, there are no problems when changing filament (M600) or when pressing the pause.

elroger, 2022-03-26:
[image]
The same thing happens to me too, I restarted the printer and it didn't read the pendrive.  Then I removed the flash drive and put it back in and if I read it at that time, later when I tried to resume printing it printed on the air with the same error appearing.  I have the same printer and the same context as the fellow in the first post.

InfinityWorldTandil, 2022-11-08:
> Hola, he observado el mismo problema. Me di cuenta cuando ocurre este problema que cuando empiezo a imprimir, la posición Z-cero no es cero pero tiene un valor muy alto sin sentido. Soluciono esto desconectando la impresora (Sidewinder X1) de la fuente de alimentación antes de imprimir. Si la posición Z-cero es entonces cero cuando se reinicia la impresión, no hay problemas al cambiar el filamento (M600) o al presionar la pausa.

Hola, tengo el mismo problema. Como sabes que la posicion de Z es cero?