wgcv/RAWR-TFT-Firmware-Artillery3D #34: [BUG] Okay/Confirmation Infinity Loop, when Filament Runs out

windows2000bug, 2022-02-19:
### Description

I am testing Fil sensor run out and it is failing with the TFT28.  

### Steps to reproduce

1. I pull out the test piece out of the fil sensor to trigger the M600. The head parks.
2. Put test filament back into fil sensor and press okay on screen. The extruder starts to run for a second or two.
3. Get okay prompts, and it says that it is being paused. This keeps happening. It seems I am locked out and have to restart the printer.

**Expected behavior**
Print Resumes

**Actual behavior**
Infinity Loop

### Hardware Variant
Stock TFT - Fil Sensor Disconnected from TFT

### TFT Firmware Version & Main Board Firmware details
Artillery X1

SKR 1.4T Board w/ Marlin 2.x - Fil Sensor Directly plugged into Main board.
OctoPrint w/ recommended M600 Firmware

### Additional Information

I did the set fil_runout to 0 as the comments for this section states for fil sensors directly connected to TFT. I assume I found a bug or the comment/documentation is wrong.

#--------------------------------------------------------------------
# Filament Runout Settings (if connected to TFT Controller)
#--------------------------------------------------------------------

#### Default FIlament Sensor
# Options: [NONE: 0, Normal: 1, SMART: 2]
fil_runout:0

* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

someitalian123, 2022-03-28:
Not much to add other than I too am experiencing this same bug. Willing to post any logs, or configs as you let me know what you need.

wgcv, 2022-03-28:
I supposed that the filament sensor is working fine. So you can try to press the continue button (Play button, sometime only the first time need to press continue) or disable the filament sensor in the configuration menu.