wgcv/RAWR-TFT-Firmware-Artillery3D #35: [Q] (Your Question?) problem too long extrusion prevented

Batouzay, 2022-03-20:
Hello, I have a problem in the middle of printing or really in a certain number of hours a message appears saying " too long extrusion preventedand " then it stops printing, what problem could it be?

my Machine is Artillery x1

jpmolina80, 2022-03-25:
I get the same error in my Artillery X1 after some hours, but in my case the print continues without issues.

Batouzay, 2022-03-25:
if that is correct, it happened to me in a particular impression, I think it was an error in the g code since before the previous message mentioned, I managed to see an error in the code, it had never happened to me and I really do not know if the g code I had problems, but since then it hasn't happened again, sorry for asking, what version of Marlin do you have installed?
Greetings and thanks for answering.

wgcv, 2022-03-28:
Hi, what slicer are you using? all are using cura? 👀

Batouzay, 2022-03-28:
If that is correct, I use a cure, will that be the reason?

wgcv, 2022-03-28:
I'm not sure why happen with Cura you could try with PrusaSlicer (Work great and my personal opinion slice better, but i like the menu of cura) But I need to check or help the community to know why Cura has the problem.

Batouzay, 2022-03-28:
Thank you friend for your time, soon I will try with prusa and I will comment again to give feedback on the result.